envirover/SPLRadioRoom #86: Update mavlink as a submodule

Davidsastresas, 2020-05-11:
I found that there was a lot of compiler warnings due to mavlink headers. I saw it has a mavlink version of 3 years ago. I've updated my local versions to use mavlink as a submodule, linked to the ardupilot/mavlink. I wonder if this may be of interest for the project. If it would be let me know so I can make a PR. 

As ardupilot/mavlink generates mavlink in the ardupilot build process, we would need to do the same here, or to fork a envirover/mavlink and generate there the headers and take that as remote ( this is what I've done in my local branches ). 

I tought I might to answer prior to doing the PR in case you don't see this valuable.

Thanks, best regards!

pavelbobov, 2020-05-12:
Hello David,

I do see your suggestion valuable. Please make the PR. I was just thinking
about updating the MAVLink library to support the MAVLink camera protocol.

I have not yet made my mind if it' better to use submodule or include the
code directly. The same MAVLink version should be used by C++ radioroom and
java UV Hub server and I'm not sure if the submodule approach will work for
the java apps (does the   java MAVLink repo even exist?).

Regards,
Pavel

envirover, 2020-05-12:
I made up my mind. submodule is the way to go.

Please make the PR into 'v2.4.0' branch.

Davidsastresas, 2020-05-12:
Great, I am on it. The mavlink repo has a python script, mavgenerate.py which is used to generate mavlink headers in any language, c, python, java, etc. I have very little knowledge about java, but I guess the java headers generated this way should be applicable to your UV Hub server. Thanks!

Davidsastresas, 2020-05-13:
I am looking at more parts of your code and I am not sure if updating mavink is going to broke, because I really haven't taken a look at UV Hub server. In mavlinktcp.cc, in function receive_message, first of all it is checked if the first byte is the mavlink_stx, but that is only the V1 stx. We should check there for both V1 and V2 stx. Also after that there are some movements in the buffers that I think are necessary for UV Hub, but I am not sure. I've had to remove them for me to work, although I am using it to connect directly to the GCS, with no UV Hub in between.

I can do the submodule PR but I will not be able to test it in the original setup because I don't have the UV Hub server setup to do so. If this is all right I will do it right away! Best regards.

envirover, 2020-05-13:
@Davidsastresas Please submit the PR into 'v2.4.0' branch. I'll update the code to support MAVLink v2 and test it with UV Hub.

closed 2021-02-24 by envirover