envirover/SPLRadioRoom #114: Add support for TCP/IP autopilot connection

envirover, 2021-04-04:
Adding support for TCP autopilot connection looks pretty straightforward. You can try it if you have time. Here are a few tips:
- RadioRoom already has class MAVLinkTCP in mavio library that works as a TCP client for MAVlink protocol. It probably could be used as-is to communicate with the autopilot.
- MAVLinkAutopilot class has field 'MAVLinkSerial serial'. We can inherit MAVLinkTCP and MAVLinkSerial from a common abstract base class (interface), called say MAVLinkInterface and  replace 'MAVLinkSerial serial' field by generic 'MAVLinkInterface autopilot'
- Add new init2(host, port, ...) method to MAVLinkAutopilot class that will initialize MAVLinkTCP and assign it to 'autopilot' field. 
- Call either MAVLinkAutopilot:init or MAVLinkAutopilot:init2 in MAVLinkHandler depending on autopilot/protocol configuration property and get TCP host and port from [autopilot] section in radioroom.conf.

gab933, 2021-06-01:
Hi Pavel,
this is a very interesting topic. Have you implemented it? I'll try also to do it by following your suggestions.
Best Regards,

Gabriele

envirover, 2021-06-02:
Hi Gabriele, I have not implemented it yet. I'll look into this after 2.5.0 release, if you will not get there first.

gab933, 2021-06-18:
Hi Pavel,
I started the implementation for TCP/UDP connection. So far, for semplicity, I just replaced the MAVLinkSerial with MAVLinkTCP (instead of merge them in a unique MAVLinkInterface class, as you suggest).
Actually I'm using UDP connection (it is just a parameter change in socket definition), and I'm ignoring the connection with UvHub.
I have mavlink connection issue (both for UDP/TCP)!
I'm able to create socket udp:0.0.0.0:14550, but it looks no stream is received in detect_autopilot->request_autopilot_version->receive_message(msg).
I tried with a mavproxy to check if I'm able to catch the mavlink stream and it works (wihtout RadioRoom I mean).
I'm investigating on socket parameters tuning.
Let's keep in contact,
Best Regards,

Gabriele

pavelbobov, 2021-06-18:
Hi Gabriele,

My experience with UDP is very limited, so I don't have any specific advice
for you.

Instead of modifying radioroom, I recommend adding MAVLinkUDP class and a
dedicated test for it (see
https://github.com/envirover/SPLRadioRoom/blob/master/tests/MAVLinkAutopilotTest.cc
for an example). This is easier than dealing with the whole stack,  and
later the test could be used to troubleshoot UDP connections.

Regards,
Pavel

gab933, 2021-06-22:
Hi Pavel,
I succeeded in the implementation. Now I arrange better the code and i make pull request.
Regards,

Gabriele